purduesigbots/pros-vsc #2: Welcome View

kunwarsahni01, 2021-05-15:
Current Status of the Welcome View:
[image]
1. Logo needs to be changed to the SVG version
2. Buttons for present on startup and analytics need to be added
3. CLI and Extension Version handlers need to be added

WillXuCodes, 2021-06-14:
Note: Logo can't be SVG now as per #10

baylessj, 2021-06-23:
Beta version in #18 

Future improvements from discord convo:

- [ ] Sticky Sidebar, scroll the middle content only
- [ ] Quick Start buttons for new project and open project on sidebar
- [ ] CLI on PATH detection + install prompt

kunwarsahni01, 2021-12-07:
This was completed.

closed 2021-12-07 by kunwarsahni01