purduesigbots/pros-vsc #4: Commands

kunwarsahni01, 2021-05-15:
The existing commands in the Tree View have all been created in the package.json and they are linked to the tree view. The linking of the commands to the middleware functions must now be done.

baylessj, 2021-05-23:
A couple todo items on uploading:

- [ ] how to have the user change the project name for upload? We can't expand to optional fields like before
- [ ] should we still assume cortex use? don't ask for slot if so

HotelCalifornia, 2021-05-23:
We can I think add a custom editor for project.pros files, akin to how VS code's settings windows looks. This way people will be able to easily change per-project upload options like remote name and slot number

baylessj, 2021-05-23:
Are you thinking of adding custom settings for the extension that would be in the native VSCode settings menu or a webview for modifying the project.pros file directly?

HotelCalifornia, 2021-05-23:
Well setting aside the part where I don't think it needs to be an entire webview (there's a whole like custom editor API), I'm not sure that we can tell vs code native settings to pull from a specific file. Also given that these settings don't translate to multi-project (workspace) or global levels like vs code native settings do, I think it makes sense not to even try doing that.

if any of that makes sense

baylessj, 2021-05-23:
Yeah, I think I'm following that. I didn't realize that there's a specific custom editor API that we can use, that definitely seems better than a custom webview. I'll take a look at that as an option.

Also do we want each PROS project to be its own VSCode workspace? or leave that up to the user to set up?

HotelCalifornia, 2021-05-23:
that's a good question, I'm not sure. my inclination is to allow multiple projects in a workspace for flexibility but I'm not sure how many people would actually use it...

I guess I could see a scenario though in which I want to work on liblvgl and the kernel side by side in the same window

we'd also then have to add some way of detecting which project the user wants to do an operation on... Maybe that's a good place for using some of that limited user input capability

baylessj, 2021-05-24:
We could handle it the way that the CMake extension does and operate on the project that contains the currently focused window but I'm not sure how difficult that would be.

My approach so far at work has been to set each of our projects as their own workspace and, if I need to look at two copies of the project, I'll open up a second VSCode window with the other workspace. This solution isn't nearly as nice as keeping both projects in the same window but is pretty manageable IMO

baylessj, 2021-05-24:
Also I'm assuming that this is the VSCode feature you were mentioning before? This is also the webview thing I was thinking of but yeah we could use it to make a nice editor for the project.pros file

https://code.visualstudio.com/api/extension-guides/custom-editors

HotelCalifornia, 2021-05-24:
> We could handle it the way that the CMake extension does and operate on the project that contains the currently focused window but I'm not sure how difficult that would be.

this is what the current pros for atom plugin does, and it includes a bunch of weird heuristics that Elliot came up with. Honestly I think that the number of people working on multiple projects at once in this way will be few enough that we can probably just add a user input step before running the command to select the project, and I don't think that will be too disruptive. If we can leverage whatever vs code uses to cache the last selected result as well I think it'll work just fine

HotelCalifornia, 2021-05-24:
> Also I'm assuming that this is the VSCode feature you were mentioning before? This is also the webview thing I was thinking of but yeah we could use it to make a nice editor for the project.pros file
> 
> https://code.visualstudio.com/api/extension-guides/custom-editors

Yeah that's what I was thinking of. Would be cool if we can replicate a view similar to vs code settings but as a fallback we can just tell vs code to parse .pros files as json and provide a schema for it for static validation

PR #5 (2021-05-25): Basic Commands
Using interactive UI with VSCode's very limited user input options was proving to be an issue, so I'm completely bypassing the middleware at the moment in favor of direct CLI calls. Not sure if that's the best approach really, but it seems to work for now....

Added three customized commands:
- Uploading
- Upgrading
- Creating a new project

All of the other user actions seem to work fine as terminal calls.

Closes #4
PR #6 (2021-06-10): Custom Editor for project.pros
[image]

Allows the user to more easily set the project's name and upload slot when modifying project.pros.

Closes #4.
closed 2021-06-10 by baylessj