purduesigbots/pros-vsc #10: Prepare for Store Release

kunwarsahni01, 2021-06-14:
Need to follow VS Extension Store Rules (cannot have svgs). See more [here](https://code.visualstudio.com/api/working-with-extensions/publishing-extension).

kunwarsahni01, 2021-06-14:
Will also need a CLI release before first extension release.

WillXuCodes, 2021-06-14:
Other than these 2 things to do before store release, what else is there to do in terms of rules?

kunwarsahni01, 2021-06-15:
Need to setup README, CHANGELOG, LICENSE see [here](https://code.visualstudio.com/api/working-with-extensions/publishing-extension#marketplace-integration).

WillXuCodes, 2021-06-15:
> Need to setup README, CHANGELOG, LICENSE see [here](https://code.visualstudio.com/api/working-with-extensions/publishing-extension#marketplace-integration).

Might be able to take on some of these. 👍

kunwarsahni01, 2021-06-15:
Mostly setup, need to fill out README, [check out store branch.](https://github.com/purduesigbots/pros-vsc/tree/feature/store-release)

closed 2021-06-22 by kunwarsahni01