purduesigbots/pros-vsc #12: One Click Install

kunwarsahni01, 2021-06-15:
Allows a user to install the CLI through the extension installation. [Check out Platformio's implementation.](https://github.com/platformio/platformio-vscode-ide/tree/develop/src/installer)

HotelCalifornia, 2021-06-17:
ultimately i think the UX will be something like this:

```
PROS CLI not found on PATH! would you like to install it?
[ standalone (recommended) ] [ with python ] [ specify custom path to PROS CLI (advanced) ]
```

the first option will download the frozen packages for the platforms that support it (windows, linux, and, eventually, macos). if the current platform is none of those, we'll fall back to the python method

the second option will attempt to use a global python installation to install pros-cli with pip. i considered asking the user if they want us to install a portable version of python in the event that there is no python install detected, but i've decided i don't think it's worth doing that. instead we'll just tell them to install python if they want to do this or else to use the standalone version

the third option i'm not convinced about, but i think it might be useful for power users who for whatever reason want to specify a custom version. alternatively that third button might just be something like `[ no, let me do it myself (advanced) ]` which may be more flexible. then we could provide a custom CLI path option in the extension settings (which would also be useful for development purposes, if say you want to have the standalone version for normal use but point it at a local copy of the pros-cli repo for testing something)

HotelCalifornia, 2021-06-17:
if we do want to add the portable python install for the case when the user has none installed, i think we should be able to do so in a future update to this

HotelCalifornia, 2021-06-17:
currently my concern is where on the filesystem the standalone CLI should be placed.

i'm thinking that the user's home directory (i.e. `%USERPROFILE%\.pros` or `$HOME/.pros`) makes sense, for a few reasons:

- this location won't clash with the configuration directories click sets up (e.g. where downloaded templates and conductor config live)
- users should always have write permissions to their own home directories, so the only thing they'll feasibly need admin permissions for is installing vs code itself
- we can very easily bootstrap adding this location to the process PATH on extension activation (which, granted, is true for any location)

kunwarsahni01, 2021-06-17:
Some integration with the welcome view for the first launch with all of this logic would make sense. Check out Gitlens and Platformios "welcome" views.

HotelCalifornia, 2021-06-17:
some have suggested using the more standard `%LOCALAPPDATA%\PROS` for the windows install location, and I think that makes sense

nickmertin, 2021-06-17:
Sounds good.
- I think VSC might provide extensions with directories for things like that? Might be worth looking into
- Another use case of the custom path one might be for invoking Windows-installed PROS CLI from WSL

HotelCalifornia, 2021-06-17:
> * I think VSC might provide extensions with directories for things like that? Might be worth looking into

you know what, it looks like you're right. 

[`vscode.ExtensionContext.globalStorageUri`](https://code.visualstudio.com/api/references/vscode-api#ExtensionContext.globalStorageUri) may just solve all the problems and make the code even more platform-independent.

the one downside i can see to this method is that it will probably be harder to offer the option to put that location on global path so the CLI can be used outside the extension, but i suppose in that case a user could just install with python on the side (etc)

PR #34 (2022-01-16): ✨One-Click
This allows users to install the PROS CLI and Toolchain all inside of VSCode, this should simply the PROS install process significantly. Closes #12
closed 2022-01-16 by kunwarsahni01