purduesigbots/pros-vsc #23: PROS Menu Buttons Don't Show

Rohan-Raman, 2021-09-19:
The menu options shown in the README.md file do not show up on my installation. This is Debian, running the latest version of Visual Studio Code.
[image]
How do I fix this? I believe that pros and prosv5 are on path and running --version displays "version 3.2.2".

Thanks for any help.

BrenekH, 2021-09-19:
You're talking about these buttons correct?

[image]

Rohan-Raman, 2021-09-19:
Yes, should’ve specified

HotelCalifornia, 2021-09-19:
do they show up if you add a pros project to the workspace?

Rohan-Raman, 2021-09-19:
Oh, I get it. In order for the buttons to show up, you have to be in an actual folder (not just the welcome page). Opening vscode as a directory has the buttons show up. Thanks!

HotelCalifornia, 2021-09-19:
yeah, those are project specific options and don't make much sense to show when there's no project around

closed 2021-09-19 by HotelCalifornia