purduesigbots/pros-vsc #30: Nothing in the extension sidebar

BenBenny11, 2021-12-07:
When I clicked the extension icon after installing the PROS extension for VEX V5, nothing came up.
[image]

I also installed PROS-cli using pip install --upgrade pros-cli

kunwarsahni01, 2021-12-07:
You need to open a PROS Project first for the extension to activate.

BenBenny11, 2021-12-07:
> 

How do I create a PROS project

kunwarsahni01, 2021-12-07:
Check out our [documentation](https://pros.cs.purdue.edu/v5/cli/conductor.html#common-tasks).

BenBenny11, 2021-12-07:
> Check out our [documentation](https://pros.cs.purdue.edu/v5/cli/conductor.html#common-tasks).

How do I open the PROS file after running pros conduct new-project ./Top-5-At-Worlds

kunwarsahni01, 2021-12-07:
You can open your PROS project folder using the open folder button from inside the file menu in VSCode.

closed 2021-12-07 by kunwarsahni01