purduesigbots/pros-vsc #37: Add a way to have .pros file display as json file and not just PROS Project Settings

TimothyRou, 2022-01-19:
Add a way to have .pros file display as json file and not just PROS Project Settings

TimothyRou, 2022-01-19:
The VScode command pallet   "select language"  (Ctrl+K M)  No text editor activ at this time.

TimothyRou, 2022-01-19:
so that don't work

HotelCalifornia, 2022-01-19:
you uh have to actually have the file open

TimothyRou, 2022-01-19:
open the file like this 

https://user-images.githubusercontent.com/78456313/150237032-2dd1cb38-64a2-4cb6-b09d-e79c63de011d.mp4

HotelCalifornia, 2022-01-19:
oh right I forgot we had a custom editor. yeah fair enough

BennyBot, 2022-01-29:
You should be able to open the file in VSCode's text-editor by doing the following:
Right-click the project.pros file, choose `Open With...`, select text editor.
To make VSCode view the file as a JSON file, press Ctrl+Shift+P to open the command palette, type and select `change language mode`, and then search and select json.

closed 2022-01-29 by BennyBot