purduesigbots/pros-vsc #45: Pros Toolchain Never Installs On MacOS 12

bradocchs, 2022-02-25:
I've tried to install PROS extension on 2 different Macs running MacOS 12 (one Intel, one M1) and both are not successful in installing the PROS toolchain. VS Code shows it installing and it seems to complete, but then the extension does not work. When I restart VS Code, it says PROS is outdated and needs to install again...over and over.

Is this a known issue? Is there a fix?

Thanks!

BennyBot, 2022-02-25:
The one-click installation is sort of confusing right now, as the popups that say "installing" or "installed" just mean downloading the toolchain. Be sure to wait until you get another popup that says "_Finished extracting pros-toolchain-macos.tar.bz_". Try to install again with this information in mind. Next update the popups will be changed to one progress bar that tracks the entire installation process so there is less confusion. If this does not work for you, try downloading it manually (instructions below), or doing another installation method such as homebrew.

**Manual installation**
Here is how to manually install PROS for VSCode **ON MAC** if one-click is not working for you.

Download this file if on an intel Mac: https://github.com/purduesigbots/pros-cli/releases/download/3.2.3/pros_cli-3.2.3-macos-64bit.zip
or this file if on an M1 Mac: https://github.com/purduesigbots/pros-cli/releases/download/3.2.3/pros_cli-3.2.3-macos-arm64bit.zip
Extract whichever zip was downloaded to this directory:
$HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/

Then, download this file: https://developer.arm.com/-/media/Files/downloads/gnu-rm/10.3-2021.10/gcc-arm-none-eabi-10.3-2021.10-mac.tar.bz2
and extract it to this directory:
`$HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-macos/`

Once that is done, we will need to run the following commands to make some files executable:
`chmod 751 $HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/pros'
`chmod 751 $HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/intercept-cc'
`chmod 751 $HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/intercept-c++'
`chmod 751 $HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/lib/*.so'

Then restart VSCode and it should work.

Hope this helps. I'll keep this issue open in case you still have problems.

bradocchs, 2022-02-25:
Thanks for the quick response!

I did wait until all the notifications were finished and it turns out that both the cli and toolchain files did install, but it still does not work and when I restart VS Code, it says it needs to install PROS again.

I tried making the 4 files executable as you instructed, but no change. However, I get the feeling that it is a file permission issue. Since the extension seems to think the files have not been installed each time VS Code is opened, does that mean it can't view directory?

Any other ideas?

BennyBot, 2022-02-25:
The directories are created by VSCode, so I don’t think it is a permissions issue. 
Try checking if pros is actually working by typing `"$HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/pros" —version`. 
If you get something that looks like an error, there might be some conflict on your system. I remember having similar issues when homebrew had a required package but a different version that what is installed by pros, and something similar may be happening to you.
If you don’t understand the error send it here and I can help decipher it.

bradocchs, 2022-02-25:
Error: /usr/local/lib/libintl.8.dylib library not loaded/not found... Looks like related to CMAKE? Do I need to install that?

I don't normally do C++ development. I'm mostly a web guy.

kunwarsahni01, 2022-02-25:
Are you on an M1 or intel machine at the moment?

bradocchs, 2022-02-25:
Intel currently, but wanting both to work.

kunwarsahni01, 2022-02-25:
Ok we build binaries for both intel and m1 and they have both been tested as working, I'm going to check them out again let me report back with what I find.

bradocchs, 2022-02-25:
FYI: Here's the error on the M1 when trying to check pros -version:
`dyld[2010]: Library not loaded: /opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Python
  Referenced from: /Users/bradspencer/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/pros
  Reason: tried: '/opt/homebrew/opt/python@3.9/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file), '/System/Library/Frameworks/Python.framework/Versions/3.9/Python' (no such file)`

bradocchs, 2022-02-25:
And for completeness, the error on Intel:
`dyld[22414]: Library not loaded: /usr/local/opt/gettext/lib/libintl.8.dylib
  Referenced from: /Users/bradspencer/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/pros
  Reason: tried: '/usr/local/opt/gettext/lib/libintl.8.dylib' (no such file), '/usr/local/lib/libintl.8.dylib' (no such file), '/usr/lib/libintl.8.dylib' (no such file)`

BennyBot, 2022-02-25:
I remember receiving this error while setting some stuff up on our virtual M1 mac. I fixed it by downloading and running the python 3.9 installer from the python website.

kunwarsahni01, 2022-02-25:
That Intel error seems to be the same error as running the M1 binary on intel..., maybe the extension is downloading the wrong binary. If you wouldn't mind https://github.com/purduesigbots/pros-cli/releases/download/3.2.3/pros_cli-3.2.3-macos-64bit.zip try running this binary manually on intel.

bradocchs, 2022-02-25:
Tried that and it gives the "not a trusted developer" message. Went to Security & Privacy and allowed it anyway. Then tried running and still get the same dylib not loading error as above.

kunwarsahni01, 2022-02-25:
Alright got a new M1 binary for you, try this one on an M1 machine: https://drive.google.com/file/d/1udxaV7ncvJWzv7AFF94VakH6Y6PAjJx8/view?usp=sharing
If you see the not trusted error try running `xattr -r -d com.apple.quarantine` on the directory of the download first

bradocchs, 2022-02-25:
Tried the new M1 binary, but got the exact same library not loaded/homebrew error. Speaking of homebrew, I've not installed it on that machine and the /opt/homebrew directory it is complaining about does not exist.

HotelCalifornia, 2022-02-25:
> Tried the new M1 binary, but got the exact same library not loaded/homebrew error. Speaking of homebrew, I've not installed it on that machine and the /opt/homebrew directory it is complaining about does not exist.

the error is simply listing off the different paths it's considering as locations while it's searching for the dynamic library. you *shouldn't* need to have homebrew installed for this to work

BennyBot, 2022-03-12:
@WillXuCodes and I had an encounter today with a student who was experiencing issues similar to yours after factory resetting their mac. 
We firstly had to install homebrew and a newer python version before encountering the `libintl.8.dylib`issue. We managed to solve the this issue through homebrew with `brew install gettext` iirc.
After that, there was a new issue with openSSL module not being found. After lots of trial and error, we got the cli working by just using the old homebrew installation for pros (https://pros.cs.purdue.edu/v5/getting-started/macos.html).

Since we are experiencing many issues with dependencies, I feel that we should have one-click attempt to use the homebrew installation method to ensure that everything is there.
Any thoughts on this from anyone?

On another note, @bradocchs I believe you came by the booth and we had a discussion where you mentioned that the linux installation process worked fine. Is it possible for you to test a new iteration of the installer on linux for us? I currently don’t have access to my laptop (it is permanently dead), but I should be able to send you this new iteration after Monday if you are interested.

bradocchs, 2022-03-13:
Yes, it was good to meet you guys at State! I'd be happy to test things for you (bradspencer@covenantchristian.org).

Obviously, the ideal would be for the VS Code extension to automate all the required deps and configs for all platforms. One-click and go! If not, the website needs step-by-step instructions that any student can easily follow.

Thanks for all your help. Looking forward to using PROS next season!

bradocchs, 2022-03-14:
Monday morning update:
On the Intel Mac I had to install Homebrew to install gettext and add the PATH to get the PROS cli to run.
Now the VS Code extension sees that PROS is installed and up-to-date, but still will not create a new project folder.
Also cannot create a new project via cli with this error:
`Updating pros-mainline... ERROR - pros.cli.conductor:new_project - HTTPSConnectionPool(host='purduesigbots.github.io', port=443): Max retries exceeded with url: /pros-mainline/pros-mainline.json (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available."))
  File "/Users/runner/.local/lib/python3.9/site-packages/requests/sessions.py", line 655, in send
  File "/Users/runner/.local/lib/python3.9/site-packages/requests/adapters.py", line 514, in send
requests.exceptions.SSLError: HTTPSConnectionPool(host='purduesigbots.github.io', port=443): Max retries exceeded with url: /pros-mainline/pros-mainline.json (Caused by SSLError("Can't connect to HTTPS URL because the SSL module is not available."))`
On the M1 Mac I had to install Homebrew to install gettext and install Python 3.9.x and add the PATH to get the cli to run.
The VS Code extension does not successfully install PROS - hangs towards the end.

BTW - I don't think I mentioned that I'm using zsh instead of bash. Could this be why it is not finding things? Also, does the cli require python 3.9?

bradocchs, 2022-03-14:
Yet another Monday development: MacOS uses LibreSSL. Could that be the reason for the SSL module error?

Software is hard. ;)

HotelCalifornia, 2022-03-14:
absolutely incredible how issues we ran into 3 years ago are still issues...

cx_Freeze is supposed to work by bundling all the required dylibs together into a standalone package but there's clearly Problems with this approach

kunwarsahni01, 2022-03-15:
I don't feel too comfortable with having VSC using homebrew to install the CLI, a lot of the extension is starting to feel like hacks that are holding everything together. We need to find the real reason begin why our cx_freeze binaries are not functioning. I don't know if we want to maybe look at another frozen python package maker but maybe we could consider that.

bradocchs, 2022-03-16:
I agree with @kunwarsahni01 that identifying the core issue is the best approach. Let me know what I can do on my end to discover and test. Thanks!

BennyBot, 2022-06-15:
Closed as the one-click improvements are released.

closed 2022-06-15 by BennyBot