purduesigbots/pros-vsc #47: Windows Compiler Requirements/Prereqs

bradocchs, 2022-02-25:
In addition to using PROS in VS Code on Macs, I need to get it working on Windows. The PROS extension and files install fine, but I don't know what compilers are supported and if there are any required compiler args.

Will it work with MinGW? If so, how do I configure it correctly. I currently have MinGW-w64 installed, but getting build errors on the default new project template.

FYI - I have successfully used PROS on Windows when installing regular Visual Studio with C++ toolchain first. I was hoping to avoid having to install all that if possible.

Thanks for any advice you can give.

HotelCalifornia, 2022-02-25:
PROS projects are built using only one toolchain, the arm-none-eabi-gcc toolchain. we package a version of this with some GNU tools (like make etc) which is included when you install PROS through the VS Code extension. I'm not sure what you're trying to do with mingw and visual studio

bradocchs, 2022-02-25:
Only tried Visual Studio and MinGW because it is not working with just the VS Code PROS install. And the obvious fact I don't know what I'm doing! ;)

I took a clean install of Windows 10, installed VS Code and PROS extension. I get includePath errors and build errors. Is the configuration incorrect? Do you have a sample c_cpp_properties.json that I can copy?

I'm a noob with C++ and trying to help our VEX team get going with PROS.

Thanks!!

HotelCalifornia, 2022-02-25:
i believe all you need to do is specify `"compileCommands": ${workspaceFolder}/compile_commands.json"`. if that doesn't work i'd recommend using the clangd extension instead of the microsoft one, which should work out of the box with no extra configuration

bradocchs, 2022-02-25:
Am I just missing a step or two?

I've uninstalled all VS Code extensions except clangd and PROS. Uninstalled and reinstalled PROS.

I'm using the Create Project link to make a new V5 project. It creates the directory and downloads the libraries. I then open that directory in VS Code. If I try to Build, it gives errors and if I open main.cpp, it cannot find main.h and therefore a bunch of undeclared identifiers.

Was I supposed to add a directory to PATH for PROS?
I don't see any config files in the project folder, are there supposed to be some?
Was I supposed to run any commands from the terminal?

The Getting Started page doesn't mention anything like that.

HotelCalifornia, 2022-02-25:
the steps you're missing are explaining to me what errors you're seeing 🙂 

from what it sounds like the project is failing to build and as a result the compilation database isn't being generated, which is the thing that basically all intellisense plugins use to display meaningful information. so let's start with what errors you're getting when trying to build

bradocchs, 2022-02-25:
PROGRESS!! For fun I ran the Clean action and then Build and it built successfully! Now I have the .d and bin directories.

However, still getting: 'main.h' file not found clang(pp_file_not_found) and all the undeclared identifiers. Seems just like a missing VS Code setting, right? It doesn't know to look in /include... How do I tell it?

HotelCalifornia, 2022-02-25:
> i believe all you need to do is specify `"compileCommands": "${workspaceFolder}/compile_commands.json"`. if that doesn't work i'd recommend using the clangd extension instead of the microsoft one, which should work out of the box with no extra configuration

this

HotelCalifornia, 2022-02-25:
(to clafrify, that key goes in c_cpp_properties.json which you had asked about earlier)

bradocchs, 2022-02-25:
Ok, but I don't see compile_commands.json anywhere either? Am I supposed to create that too?

bradocchs, 2022-02-25:
Do I need to run?: cmake -DCMAKE_EXPORT_COMPILE_COMMANDS=1

HotelCalifornia, 2022-02-25:
it should be automatically created, provided you built the project using the extension buttons/commands to do so...

bradocchs, 2022-02-25:
I used the extension Build action link. It did not create that file. I tried cmake command, but cmake is not recognized.

HotelCalifornia, 2022-02-25:
yeah I've just been informed that the extension isn't running the correct command due to various issues. cmake isn't going to help you here because we don't use cmake as a build system. just bear with me for a minute

HotelCalifornia, 2022-02-26:
> So am I understanding it correctly that VS Code's syntax checker needs a compiler installed to work? If so, can we point VS Code to the compiler that's included with the PROS installation?

yes, this is what the compilation database (compile_commands.json) is for when it's generated properly.

> I hope there's a way to get it working without that. Looking forward to finding out. Thanks so much for your help!

There is, as I mentioned above. Keep an eye on #48 which should solve most of your problems.

After that's merged and released, you'll be able to use the Microsoft c/c++ extension with the c_cpp_properties.json configuration i mentioned above

closed 2022-02-28 by kunwarsahni01