purduesigbots/pros-vsc #51: Issues on Windows Persist

bradocchs, 2022-03-05:
I tried release 0.3.0 on a fresh install of Windows 11 and was able to create a project and build successfully, but as soon as I installed the MS C/C++ extension pack everything stopped working. :( I then tried uninstalling all but the PROS extension and now it won't even create a new project. Also tried uninstalling PROS (via the extension tool) and even the PROS extension...still dies before even creating the folder for a new project.

Am I using it incorrectly? Is there a PROS log I can view?

Thanks for your help!

HotelCalifornia, 2022-03-05:
can you be more specific than "everything stopped working"

bradocchs, 2022-03-05:
At first, a bunch of "red squiggles" from the MS extension - under most of the header includes and a complaint about clang config in the terminal. I permanently deleted (bad habit) the first project that worked while testing, so I don't have the exact error messages. Now the Create Project link in the PROS tool asks for a project name, but then nothing ever happens. The project directory is not created and it never asks for the version of PROS to download. Next time I try from a fresh install, I will document everything and take screenshots.

Before I added the VS Code recommended extension pack, it created a project and successfully built it. The only thing that didn't work was C++ intellisense. Why would installing extensions break all the PROS functionality?

Are there things on my end I can do to troubleshoot? Sorry I'm still fairly new to C/C++ and PROS.

bradocchs, 2022-03-06:
Just a quick update: I went ahead and did another fresh Windows install (a full wipe). This time the PROS extension won't create a project even before installing additional extensions, so the issue must be unrelated to the MS extensions I had blamed earlier. Could it be a Windows 11 issue?

I can create a project and build from the terminal, but I want it to work via the extension. Let me know if there's anything I can do to help troubleshoot on my end. Thanks.

BennyBot, 2022-06-15:
Closed as the one-click improvements are released.

closed 2022-06-15 by BennyBot