purduesigbots/pros-vsc #58: Action buttons do not work if project not open (with workaround)

MarinRobotics, 2022-05-16:
I am having a similar issue to #55 since switching from the Atom extension; for reference, I am on Windows. Everything actually works with the extension if I have a PROS project open. But when I open a blank VS Code window, the build button just builds forever (because there is no project to build, I would assume), and the upload button silently fails. There should be some error message that tells the user that there is no project open. The "Upgrade Project" button fails with the error:

> Command failed: pros c info-project --project "undefined" --machine-output undefined
> 'pros' is not recognized as an internal or external command,
> operable program or batch file.

Perhaps more importantly, creating a project fails silently for me if a project is not open. After asking for the project location, the project type (v5 or cortex), and the project name, nothing happens.

There is a workaround to some of these issues, though. If you click "Brain Terminal" in a VS Code with no PROS project open, the PROS terminal opens. After this, a window with no project still builds forever after clicking "Build Project," and "Upgrade Project" fails, but with a different error:

> Command failed: pros c info-project --project "undefined" --machine-output 
> Usage: pros c info-project [OPTIONS]
> 
> Error: Invalid value for "--project": Path "undefined" does not exist.

This makes sense because there is no project open, but perhaps the error message should be more descriptive, telling the user that there is no project open. With all that said, this workaround *does* fix project creation; however, it is still a workaround and these bugs should be fixed.

BennyBot, 2022-06-13:
Referenced this issue in #53 as it should resolve the difference between before and after clicking "Brain Terminal". We will close this once we add more descriptive error messages. Thanks for the suggestion.

BennyBot, 2022-12-13:
Closed by #121