purduesigbots/pros-vsc #105: 🐛VS-Code GUI can't run any cli commands due to a lack of sudo permissions (experienced by multiple people)

tongclement, 2022-10-25:
#### Bug Description
The VSCode PROS extension cannot run any commands like new project, build, upload, etc due to a lack of sudo commands. 
The error messages are:
```
......flywheel pros3 % pros upload
ERROR - pros.cli.main:main - [Errno 13] Permission denied: '/usr/local/Cellar/pros-cli/3.2.2/libexec/lib/python3.9/site-packages/idna-2.7-py3.9.egg-info/PKG-INFO'
  File "/usr/local/Cellar/pros-cli/3.2.2/libexec/lib/python3.9/site-packages/pkg_resources/__init__.py", line 1611, in _get
    with open(path, 'rb') as stream:
PermissionError: [Errno 13] Permission denied: '/usr/local/Cellar/pros-cli/3.2.2/libexec/lib/python3.9/site-packages/idna-2.7-py3.9.egg-info/PKG-INFO'
(base) clementtongpersonal@HISHK-MBP1210-3 flywheel pros3 % 
```
Running eg `pros upload`  doesn't work but ` sudo pros upload`  works
Experienced by multiple people
This should be an easy fix. 
If this can't be fixed, at least include it in the documentation.
Many thanks

#### Platform
Mac OS on Apple Silicon (M1)

#### Steps to reproduce
Install PROS and PROS CLI and toolchain inside VS-Code's extension installer on a Apple Silicon Mac

#### Additional Information

Refer to this discussion here: https://www.vexforum.com/t/temporally-fixed-awaiting-permanent-solution-pros-vscode-cannot-upload-code-to-brain-due-to-python-permission-error/108096

Many thanks

BennyBot, 2022-10-25:
Based on the error message, it looks like you are using an older install of the PROS CLI that was not installed through the VSCode extension. We have not had any issues with sudo on Mac or Linux devices, as the CLI should inherit whatever permissions VSCode has. 
Have you tried installing the PROS CLI and PROS toolchain through the VSCode extension? Additionally, this will require you to run terminal commands through the PROS Integrated Terminal (Accessible with the `Integrated Terminal` button in the PROS sidebar). If you don't want to use the terminal we provide in VSCode, you can configure your environment variables to support the VSCode install following [these instructions](https://github.com/purduesigbots/pros-cli/issues/192)

tongclement, 2022-10-28:
I was using version 3.3.2, installed via the VSCode extension, with commands ran inside the vscode integrated terminal. 
Problem Solved by running: `sudo chmod -R 777 /usr/local/Cellar/pros-cli/3.2.2/`

```
(base) username@macbookname prosprojectname % pros --version
ERROR - pros.cli.main:main - [Errno 13] Permission denied: '/usr/local/Cellar/pros-cli/3.2.2/libexec/lib/python3.9/site-packages/idna-2.7-py3.9.egg-info/PKG-INFO'
  File "/usr/local/Cellar/pros-cli/3.2.2/libexec/lib/python3.9/site-packages/pkg_resources/__init__.py", line 1611, in _get
    with open(path, 'rb') as stream:
PermissionError: [Errno 13] Permission denied: '/usr/local/Cellar/pros-cli/3.2.2/libexec/lib/python3.9/site-packages/idna-2.7-py3.9.egg-info/PKG-INFO'
(base) username@macbookname prosprojectname sudo chmod -R 777 /usr/local/Cellar/pros-cli/3.2.2/
Password:
(base) username@macbookname prosprojectname pros --version
pros, version 3.2.2
(base) username@macbookname prosprojectname % 
```

AndrewLuGit, 2022-10-28:
1. When you run `pros --version`, it shows that the version is 3.2.2, not 3.3.2.
2. The VSCode extension should install the CLI at `$HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/ `, not `/usr/local/Cellar/pros-cli/3.2.2/ `
I think your path to the CLI is configured incorrectly, hence why you are experiencing these issues. If you check the [extension source code](https://github.com/purduesigbots/pros-vsc/blob/2437bd6c4dbc89dc419f529db5e2421f39846274/src/one-click/install.ts#L489), it already performs the chmod operation after installing the CLI.

BennyBot, 2022-10-28:
In addition to what Andrew said, the reason your PATH configuration is incorrect is mostly likely if you are not using the PROS integrated terminal, or if you never actually installed the CLI through the VSCode extension. Below is a screenshot of how to access and confirm that you are in a PROS terminal.
[image]

If you are in a PROS terminal, please run the command `echo $PATH` and send the output so that we can further identify the issue you are having.

tongclement, 2022-10-29:
Yes - I can confirm I am running in the `PROS Terminal` inside VSCode and here is the output of echo $PATH
```
% echo $path
/usr/local/sbin /opt/homebrew/bin /opt/homebrew/sbin /opt/local/bin /opt/local/sbin /Library/Frameworks/Python.framework/Versions/3.10/bin /usr/local/bin /usr/bin /bin /usr/sbin /sbin /Library/Apple/usr/bin /Library/Frameworks/Mono.framework/Versions/Current/Commands /Users/clementtongpersonal/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos /Users/clementtongpersonal/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-macos/bin /usr/local/sbin /Users/clementtongpersonal/miniforge3/bin /Users/clementtongpersonal/miniforge3/condabin /opt/homebrew/bin /opt/homebrew/sbin /opt/local/bin /opt/local/sbin /Library/Frameworks/Python.framework/Versions/3.10/bin
```

tongclement, 2022-10-29:
My VS Code settings:
Pros: One Click: Cli Download URL
URL to download PROS CLI from. SHOULD NOT BE CHANGED BY MOST USERS!
`https://github.com/purduesigbots/pros-cli/releases/download/3.3.3/pros_cli-3.3.3-macos-arm64bit.zip`

Pros: One Click: Toolchain Download URL
URL to download PROS Toolchain from. SHOULD NOT BE CHANGED BY MOST USERS!
default

When I click PROS Update, it says `PROS is up to date`

tongclement, 2022-10-29:
I previously did install the atom pros editor but I since deleted the old CLI and reinstalled a new one

tongclement, 2022-10-29:
Many thanks for your kind help in advance

BennyBot, 2022-10-29:
Ok, can you verify that files exist at `$HOME/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-cli-macos/ `? If not, I ask that you wait for our next VSCode extension update, which is aimed at being within the next month. If there are files, I would recommend deleting the PROS install located at `/usr/local/Cellar/pros-cli/` and it should hopefully resolve the problem. If it doesn't resolve the problem, I can further assist.

BennyBot, 2022-12-13:
Any more updates to the issue now that VSCode 0.5.0 has been released? Can you please check and see if the issue still remains?

BennyBot, 2022-12-30:
Closing due to no response. Please reopen if issue still persists after updating to extension version 0.5.0.

closed 2022-12-30 by BennyBot