purduesigbots/pros-vsc #130: 🐛 Default keybinding for Integrated Terminal is the default Linux Format Document keybind

Chrono-byte, 2022-12-13:
#### Bug Description

The default keybinding for Integrated Terminal is the default Linux Format Document keybinding for VSCode.

#### Platform

Fedora Rawhide

#### Steps to reproduce

1. Stock VSCode install
2. Install PROS
3. Observe.

#### Additional Information

I'm aware that Linux is a 3rd tier userbase but it seems a little subpar to overwrite a stock keybind.

Chrono-byte, 2022-12-13:
https://github.com/purduesigbots/pros-vsc/commit/e373e3f0901cfb2d83870b96e4cee3b867b8cd54#diff-7ae45ad102eab3b6d7e7896acd08c427a9b25b346470d7bc6507b6481575d519R160

AndrewLuGit, 2022-12-13:
Do you have any suggestions for what we should change the key binding to?

Chrono-byte, 2022-12-13:
Ctrl+Alt+I would work.

BennyBot, 2023-04-12:
Fixed in #149. Looked through the VSCode keyboard shortcut reference for all platforms and these new keybinds should work for everyone.

closed 2023-04-12 by BennyBot