purduesigbots/pros-vsc #132: 🐛

markhayes2, 2022-12-14:
#### Bug Description

<p>
Can you point me in the right direction?
I was using the PROS extension in Atom for Cortex, and the toolchain installed with Homebrew.
I am now trying to get started with the PROS extension in VCS for Cortex, and getting the following in the Output panel after creating a new project.
Would like to clean up those warnings.
</p>
#### Platform

<p>
Intel/M1 mac
</p>
#### Steps to reproduce

<p>
Create a new Cortex project with VCS Pros extension
</p>
<p>
[14/12/2022, 19:22:53] Unable to resolve configuration with compilerPath "/Users/mhayes/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-macos/bin/arm-none-eabi-g++".  Using "/opt/homebrew/bin/arm-none-eabi-gcc" instead.<br><br>
[14/12/2022, 19:22:53] Unable to resolve configuration with compilerPath "/Users/mhayes/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-macos/bin/arm-none-eabi-g++".  Using "/opt/homebrew/bin/arm-none-eabi-gcc" instead.<br><br>
[14/12/2022, 19:22:53] For C source files, IntelliSenseMode was changed from "macos-gcc-arm" to "macos-clang-arm" based on compiler args and querying compilerPath: "/usr/bin/clang"
</p>

#### Additional Information

markhayes2, 2022-12-14:
> It looks to be an issue with how we configure the VSCode C/C++ extension.
> Try going to the .vscode/c_cpp_properties.json file in your project and
> changing the compilerPath to "/opt/homebrew/bin/arm-none-eabi-g++".

Many thanks Andrew, I will try that.
- Mark

markhayes2, 2022-12-16:
Changing the compilerPath setting did not seem to fix it.
I uninstalled the extension and reinstalled, and it seems ok now.
The reinstall set compilerPath to this:
"compilerPath": "/Users/mhayes/Library/Application Support/Code/User/globalStorage/sigbots.pros/install/pros-toolchain-macos/bin/arm-none-eabi-g++",

AndrewLuGit, 2022-12-16:
Glad to hear. If you haven't already, you should probably also uninstall the arm toolchain you installed through homebrew.

closed 2022-12-16 by AndrewLuGit