purduesigbots/pros-vsc #134: 🐛

ifndef-define, 2023-01-07:
#### Bug Description
The `Run` quick action command got stuck running the first program I compiled and only runs that program. After deleting that specific program from the brain, it returns this error;
```
ERROR - __main__:main - Device NACK'd with reason: Program file error
Sentry is attempting to send 1 pending error messages
Waiting up to 2 seconds
Press Ctrl-Break to quit
```
I have tried closing all open projects, and reloading PROS, VSCode, and the brain, and nothing works.
#### Platform
Windows 11 with VSCode 

#### Steps to reproduce
Run the `Build and Upload` command in the quick action toolbar for one program and then use the `Run` command.
Run the same command for another program in a different slot, and run. I observed the first program ran instead.
Delete the first program, and the `Run` command fails.

#### Additional Information
n/a

BennyBot, 2023-02-02:
The problem from what I can tell is that it always tries to run the program in program slot 1 on the brain.

I think the solution here is going to be one of the following:
1) The user is prompted to pick a program slot from 1-8
2) VSCode chooses the program based on upload_options from the project.pros file

Which would you prefer?

ifndef-define, 2023-02-02:
The second option is where it reads from the project.pros file would be most convenient. That being said, the first option could be used in the case that there is no pros project open (just the extension is running).

Given the fact the command line exists, it only makes sense to have option 2 in play, since I could just use the command line should I want to run other programs.

PR #146 (2023-04-12): 🐛 Detect Slot for Run Command
Read `project.pros` file to find the slot for the program, which is then used for the Run command.
Closes #134
closed 2023-04-12 by BennyBot