purduesigbots/pros-vsc #158: The pros plugin in vscode 1.79.2 cann't create a project with the default project structure.

peaceyear, 2023-06-23:
#### Expected Behavior:

a project with the structure like https://pros.cs.purdue.edu/v5/tutorials/walkthrough/clawbot.html.
project
│   project.pros        (used by PROS CLI to know kernel version and other metadata)
│   Makefile            (instructs make how to compile your project)
|   common.mk           (helper file for Makefile)
│
└───src                 (source files should go here)
│   │   main.cpp        (source for competition task functions, like operator control and autonomous)
|
└───include             (Header files should go in here)
│   │   api.h           (Lets source files know PROS API functions)
│   │   main.h          (Includes api.h and anything else you want to include project-wide)
|   └───pros            (Contains all of the specific header files for the PROS API functions)
|   └───okapi           (Contains all of the header files for OkapiLib)
|   └───display         (Contains all of the header files for LVGL, the graphics library for the V5 screen)
│
└───firmware
│   libpros.a       (Pre-compiled PROS library)
│   okapilib.a      (Pre-compiled OkapiLib library)
|   v5.ld           (Instructs the linker how to construct binaries for the V5)

#### Actual Behavior:

an empty project.

#### Steps to reproduce:

1. open the vscode with the pros plugin installed.
2. click the pros plugin icon on the left side bar.
3. click the Create Project menu on the left pros plugin side bar.
4. select one folder for the project
5. use the default input to create project
6. get an empty project.

#### System information:
Platform:  v5
PROS Kernel Version:  3.8.0

#### Additional Information

shall I connect the robotic before I create the project?

#### Screenshots/Output Dumps/Stack Traces
[image]

BennyBot, 2023-06-23:
This is a known problem that we are actively working on. For now, create the project using the PROS Integrated Terminal. Run the command `pros c n your-project-name-here`, and then open the created project using VSCode File -> Open Folder -> Your Project. If you receive the error `pros not recognized` or `pros not found` or something similar, this means you either have the wrong terminal open (click the Integrated Terminal button in the PROS Sidebar), or you do not have PROS installed (click Install PROS in the PROS Sidebar)

peaceyear, 2023-07-02:
Thanks for BennyBot. The integrated terminal works well.

closed 2023-09-03 by BennyBot