purduesigbots/pros-vsc #172: 🐛vs extension not installing ok, toochain not working??

vex-lq, 2023-08-22:
🐛vs extension not installing ok, toochain not working

BennyBot, 2023-08-22:
It would be very helpful if you could provide some screenshots of the issue. We are unable to help with the information you have given us.

vex-lq, 2023-08-22:
cli seems to be installed, the popup showed that info

BennyBot, 2023-08-22:
ok, we will do manual installation. I'm assuming this is windows based on the other issue. Do the following steps:

1) close vscode
2) open file folder and navigate to `%appdata%\Code\User\globalStorage\sigbots.pros\install\`
3) create a new folder called `pros-toolchain-windows` in that location
4) download the following toolchain: https://github.com/purduesigbots/toolchain/releases/download/v10.3-2021.07/pros-toolchain-windows-formatted.zip
5) extract the contents of that zip folder into the folder created in step 2

after these steps are done, your file explorer should look like this:
[image]

6) open vscode and it should verify the toolchain as working.

let me know if you have more problems.

vex-lq, 2023-08-22:
[image]
[image]
[image]
still problem..

vex-lq, 2023-08-22:
waaa...it's solved, I unzipped a standalone toolchain at other dir and added it to sys environment.  now all that is deleted,   vscode verifies the toolchain as working.  i have upload a project to Brain.
thx! @BennyBot

vex-lq, 2023-08-22:
[image]
when it's running code, i click the stop、run button it pops this..

and another question, i can not find std::cout output contents, and what's the difference bettwen brain terminal and intergrated terminal?  @BennyBot

BennyBot, 2023-08-22:
> and another question, i can not find std::cout output contents, and what's the difference bettwen brain terminal and intergrated terminal? @BennyBot

Integrated Terminal starts a terminal with the correct environment variable configuration for the vscode installed PROS cli and toolchain. Brain Terminal spawns an integrated terminal and automatically runs the command `pros terminal`, which opens a serial communication connection to the connected v5. Brain Terminal is what you want to use to view the cout output contents.

BennyBot, 2023-08-22:
The port error is interesting since it was able to upload successfully. It looks like it is unable to access the device. I am not sure how to immediately fix it, often times it will fix itself given time.

vex-lq, 2023-08-23:
when i restart vscode, and no ther ides are running, i can upload a project, then i click run button, all is ok so far, but when i click stop button, error below pops up..
[image]
frustrating.. @BennyBot

BennyBot, 2023-08-25:
interesting. I will look into this later today.