purduesigbots/pros-vsc #186: Integration of PROS VSCode Extension with Remote-SSH and Local 'Brain' Support?

yuedai, 2023-09-20:
We are newcomers to the world of PROS, and we're excited to discover that PROS has an excellent Visual Studio Code (VSCode) extension. I'm eager to find out if the PROS VSCode extension can seamlessly integrate with Remote-SSH.

Given that C++ compilation can be quite resource-intensive, we're planning to set up a more robust workstation that allows team members to build their C++ code efficiently. I've already experimented with the Remote-SSH extension in conjunction with the PROS extension, and so far, they seem to work well for the build and clean processes. However, our team members might find themselves in remote locations, needing to connect their Brain device to their laptops but still utilizing the powerful workstation for code compilation.

Currently, I don't have a physical 'Brain' device on hand. Before we proceed further, I'd like to inquire **whether the PROS VSCode extension supports a local 'Brain' device within a Remote-SSH build environment.**

noam987, 2023-09-20:
Have build times and resources usage been an issue that you've encountered or is this a hypothetical situation? Build times are usually pretty quick, even on low powered devices such as Raspberry Pis.

yuedai, 2023-09-20:
I didn't try Raspberry PI but with our poor laptop the compilation time is a little long. We can leverage/borrow one remote workstation with little/no cost, so I am trying to figure out if local Brain + Remote-SSH build is doable.

BennyBot, 2023-09-21:
So from what I'm understanding, your current process works like this:
1) Student opens VSCode on their laptop and connects to workstation through Remote-SSH
2) Student opens PROS project and extension from within that Remote-SSH (Extension and project are being saved onto the workstation, not the user's laptop)
3) Student codes and then builds the project on the workstation
4) Student will connect the robot to the workstation and click upload through the Remote-SSH window, prompting the workstation to download the project onto the robot.
This process above seems fine, granted the students' and their robot(s) are able to access the workstation.

To answer your question about support for downloading the project onto a brain connected to the student's laptop when the project is saved on the workstation (assuming your current process is what I've detailed above), our extension currently (and very likely never will) support this. That would require our CLI to somehow figure out that SSH connection, and download the compiled binaries from the workstation back onto the laptop. That would require a lot of work on our end across multiple repositories for something that should never need to be done in the first place.

I agree with what Noam said. I've never experienced problems or time delays compiling PROS projects, even on low power devices like a Raspberry Pi Zero. If the student's laptop is able to run VSCode, it should be more than capable of compiling a PROS project in a reasonable amount of time. Perhaps we have different definitions of what is a reasonable compile time. 

If you really want to stick with the remote build, you will just need to find a way for the user to build remotely, and then fetch the compiled binaries onto their local machine. However, I'm sure you will find that this process will take much longer than just building locally.

yuedai, 2023-09-21:
Thank you @BennyBot for your explanation. I understand your design We will follow the standard practice, that put everything in our laptop.

However, in the end, I plan to tinker a bit and see if I can use usbip/usbipd + ssh to map my local Brain device to the Workstation. I feel like this should be relatively straightforward, just need to prepare some scripts.

closed 2023-09-21 by BennyBot