purduesigbots/pros-vsc #206: PROS CLI and TOOLCHAIN issue, I do not know how to fix it. 🐛 please send help.

lizards777, 2024-02-02:
#### Bug Description

[image]

#### Platform (vs code)

#### Steps to reproduce

i dont know how to fix

apparently, the CLI and TOOLCHAIN are outdated or just not working in general.

Rocky14683, 2024-02-02:
You are probably not on PROS integrated terminal.

lizards777, 2024-02-02:
i think i am on it
[image]

lizards777, 2024-02-02:
pls help

lizards777, 2024-02-02:
[image]
I am getting this problem.
Then when I press "Install now!", it gives me the same thing as before.
[image]

AndrewLuGit, 2024-02-02:
Could you go to `%appdata%\Code\User\globalStorage\sigbots.pros\logs` and send us the log file for when the installation fails?

lizards777, 2024-02-02:
[PROS_Extension_log_2024-02-02T17-33-08.647Z.txt](https://github.com/purduesigbots/pros-vsc/files/14143515/PROS_Extension_log_2024-02-02T17-33-08.647Z.txt)

lizards777, 2024-02-02:
> Could you go to `%appdata%\Code\User\globalStorage\sigbots.pros\logs` and send us the log file for when the installation fails?

[PROS_Extension_log_2024-02-02T17-33-08.647Z.txt](https://github.com/purduesigbots/pros-vsc/files/14143532/PROS_Extension_log_2024-02-02T17-33-08.647Z.txt)

AndrewLuGit, 2024-02-02:
Looks like the error is here:
```
OneClick | 2024-02-02T17:38:04.248Z | ERROR :: Error: Command failed: pros c --help --machine-output 
This version of c:\Users\ecrro\AppData\Roaming\Code\User\globalStorage\sigbots.pros\install\pros-cli-windows\pros.exe is not compatible with the version of Windows you're running. Check your computer's system information and then contact the software publisher.
```
What version of Windows are you running? And if it's Windows 10/11, could you find your device specifications and check what type of processor your computer has?

lizards777, 2024-02-02:
[image]
[image]
I don't know, I should have all the specs to run the PROS.

AndrewLuGit, 2024-02-02:
From what I could find online, it might also be due to a corrupted download. Maybe try uninstalling PROS and then reinstalling.

lizards777, 2024-02-02:
> From what I could find online, it might also be due to a corrupted download. Maybe try uninstalling PROS and then reinstalling.
[image]
I reinstalled the PROS along with the default CLI and Toolchain files and am still getting the same error.
Also, I have tried downloading the files manually which didn't work. We also tried factory resetting and trying to download everything again on multiple computers but kept getting the same error.

lizards777, 2024-02-02:
All of the computers were windows 10 and Lenovo laptops. I don't know if that changes anything.

AndrewLuGit, 2024-02-02:
Maybe try running the Windows program compatibility troubleshooter https://support.microsoft.com/en-us/windows/make-older-apps-or-programs-compatible-with-windows-783d6dd7-b439-bdb0-0490-54eea0f45938 If that doesn't work, I'm not sure what else you can do.

AndrewLuGit, 2024-02-02:
I think the download links must have broken somehow, I am also getting this on my computer

lizards777, 2024-02-02:
> I think the download links must have broken somehow, I am also getting this on my computer

Yeah, it is weird because I have tried with multiple computers and windows 11 and 10 computers and the default links that the PROS extension installs keep repeating the same error.

noam987, 2024-02-02:
Something broke yesterday. We are looking into it

lizards777, 2024-02-02:
OMG THEY FIXED IT. THE PROS V0.7.1 UPDATE FIXED THE BUG. THANK YOU SO MUCH!!!!

AndrewLuGit, 2024-02-02:
I’m glad that worked. Thank you for bringing this to our attention as well!

closed 2024-02-02 by AndrewLuGit