purduesigbots/pros-vsc #210: Cli and Toolchain not working. Install failed.

gonzodog22, 2024-02-03:
Whenever Pros is open, it tries to verify the installation. It fails after I try to install it. I already tried to uninstall and reinstall both the extension and vsc. I tried manually uploading and extracting the toolchain myself but still no dice.

BennyBot, 2024-02-03:
1) What platform is this on? (windows, mac, linux)
2) Make sure you have updated our extension to version 0.7.1

gonzodog22, 2024-02-25:
This is on windows and i reinstalled the extension to the latest version. This is the error: Unable to resolve configuration with compilerPath: "c:\Users\Owner\AppData\Roaming\Code\User\globalStorage\sigbots.pros\install\pros-toolchain-windows\usr\bin\arm-none-eabi-g++"

Just-Do-It-G, 2025-04-30:
> This is on windows and i reinstalled the extension to the latest version. Things seem to be working fine but this error displays in the output: Unable to resolve configuration with compilerPath: "c:\Users\Owner\AppData\Roaming\Code\User\globalStorage\sigbots.pros\install\pros-toolchain-windows\usr\bin\arm-none-eabi-g++".

Did u ever solve this issue?

gonzodog22, 2025-04-30:
Yea I fixed it a while ago by downgrading versions. Use the pros c upgrade command to do this

Just-Do-It-G, 2025-04-30:
How many versions did u downgrade.