purduesigbots/pros-vsc #222: 🐛Create Project fails for V5 with "Unterminated string in JSON at position 8182"

arrowsmith001, 2024-05-17:
#### Bug Description

User cannot create a new PROS V5 project through the PROS VS Code extension. This seems to happen on Mac OS, but not Windows.

#### Platform

MacBook Pro running Sonoma 14.2.1

#### Steps to reproduce

1. Install the PROS extension for VS Code
2. Install PROS CLI and Toolchain through the extension
3. Through the extension menu: PROS -> Conductor -> Create Project
4. Select a valid destination folder
5. Enter a valid project name
6. Select v5
7. "Fetching kernel versions..." will pop up
8. Then, "Command completed successfully!" followed by the error, as shown

[image]

At this point the PROS extension commands seem unresponsive.

Jackman3323, 2024-05-18:
@arrowsmith001 what version of pros kernel were you attempting to use? Or did you never get to choose?

Lunar-Eclipse255, 2024-05-18:
I am having the same exact issue, when creating a project it doesn't ask for a kernel version like it has in previous versions, but I am on kernel version 4.0.5. This is the full error message I get in PROS output: Uc&42BWAaQ{"type": "log/message", "level": "INFO", "message": "INFO -
pros.cli.common:callback - Debugging messages enabled - pros-cli
version:3.5.1", "simpleMessage": "Debugging messages enabled"}
Uc&42BWAaQ{"text": "Not sending analytics for this command.\n\n",
"type": "notify/echo", "notify_value": 0}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- pros:callback - CLI Version: 3.5.1 - pros-cli version:3.5.1",
"simpleMessage": "CLI Version: 3.5.1"}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- pros.cli.common:callback - None@None (v5, BaseTemplate) - pros-cli
version:3.5.1", "simpleMessage": "None@None (v5, BaseTemplate)"}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- pros.config.config:__init__ - Opening
/Users/gagrehansavona/Library/Application Support/PROS/conductor.pros
(Conductor) - pros-cli version:3.5.1", "simpleMessage": "Opening
/Users/gagrehansavona/Library/Application Support/PROS/conductor.pros
(Conductor)"}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- pros.config.config:__init__ - Opening
/Users/gagrehansavona/Library/Application Support/PROS/cli.pros
(CliConfig) - pros-cli version:3.5.1", "simpleMessage": "Opening
/Users/gagrehansavona/Library/Application Support/PROS/cli.pros
(CliConfig)"}
Uc&42BWAaQ{"type": "log/message", "level": "INFO", "message": "INFO -
pros.conductor.depots.depot:get_remote_templates - Last check of
LemLib was 2024-05-17 21:29:01.300209 (0:02:14.642745 vs 0:01:00). -
pros-cli version:3.5.1", "simpleMessage": "Last check of LemLib was
2024-05-17 21:29:01.300209 (0:02:14.642745 vs 0:01:00)."}
Uc&42BWAaQ{"text": "Updating LemLib... ", "type": "notify/echo",
"notify_value": 1}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- urllib3.connectionpool:_new_conn - Starting new HTTPS connection
(1): raw.githubusercontent.com:443 - pros-cli version:3.5.1",
"simpleMessage": "Starting new HTTPS connection (1):
raw.githubusercontent.com:443"}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- urllib3.connectionpool:_make_request -
https://raw.githubusercontent.com:443 \"GET
/LemLib/LemLib/depot/stable.json HTTP/1.1\" 200 393 - pros-cli
version:3.5.1", "simpleMessage":
"https://raw.githubusercontent.com:443 \"GET
/LemLib/LemLib/depot/stable.json HTTP/1.1\" 200 393"}
Uc&42BWAaQ{"text": "Done\n", "type": "notify/echo", "notify_value": 1}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- pros.config.config:__init__ - Opening
/Users/gagrehansavona/Library/Application Support/PROS/cli.pros
(CliConfig) - pros-cli version:3.5.1", "simpleMessage": "Opening
/Users/gagrehansavona/Library/Application Support/PROS/cli.pros
(CliConfig)"}
Uc&42BWAaQ{"type": "log/message", "level": "INFO", "message": "INFO -
pros.conductor.depots.depot:get_remote_templates - Last check of
pros-mainline was 2024-05-17 21:29:01.683101 (0:02:14.581676 vs
0:01:00). - pros-cli version:3.5.1", "simpleMessage": "Last check of
pros-mainline was 2024-05-17 21:29:01.683101 (0:02:14.581676 vs
0:01:00)."}
Uc&42BWAaQ{"text": "Updating pros-mainline... ", "type":
"notify/echo", "notify_value": 2}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- urllib3.connectionpool:_new_conn - Starting new HTTPS connection
(1): pros.cs.purdue.edu:443 - pros-cli version:3.5.1",
"simpleMessage": "Starting new HTTPS connection (1):
pros.cs.purdue.edu:443"}
Uc&42BWAaQ{"type": "log/message", "level": "DEBUG", "message": "DEBUG
- urllib3.connectionpool:_make_request -
https://pros.cs.purdue.edu:443 \"GET
/v5/_static/releases/pros-mainline.json HTTP/1.1\" 200 20297 -
pros-cli version:3.5.1", "simpleMessage":
"https://pros.cs.purdue.edu:443 \"GET
/v5/_static/releases/pros-mainline.json HTTP/1.1\" 200 20297"}
Uc&42BWAaQ{"text": "Done\n", "type": "notify/echo", "notify_value": 2}
[... 136 lines cut ...]
'local', 'target': 'v5', 'local': True}\n{'name': 'LemLib', 'version':
'0.5.0', 'location': 'LemLib', 'target': 'v5', 'local':
True}\n{'name': 'LemLib', 'version': '0.4.9', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.4.8', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'LemLib', 'version': '0.4.7', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.4.6', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'LemLib', 'version': '0.4.5', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.4.4', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'LemLib', 'version': '0.4.3', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.4.2', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'LemLib', 'version': '0.4.1', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.4.0', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'LemLib', 'version': '0.3.3', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.3.2', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'LemLib', 'version': '0.3.1', 'location': 'LemLib',
'target': 'v5', 'local': False}\n{'name': 'LemLib', 'version':
'0.3.0', 'location': 'LemLib', 'target': 'v5', 'local':
False}\n{'name': 'ARMS', 'version': '3.2.0', 'location': 'local',
'target': 'v5', 'local': True}"}

arrowsmith001, 2024-05-18:
> @arrowsmith001 what version of pros kernel were you attempting to use? Or did you never get to choose?

Same as @Lunar-Eclipse255, I didn't get as far as choosing the PROS kernel version. It seems to fail while fetching them.

Jackman3323, 2024-05-21:
@arrowsmith001, @Lunar-Eclipse255: Just an update for y'all while we look into _how_ the pros conductor file is breaking in this way, running pros c reset should get you up and running again. It will reset your locally cached templates/depots to the default so things like lemlib will need to be re-fetched. The reset command will be added as a button in the next update (see #224). Please let us know here if this doesn't fix project creation.

arrowsmith001, 2024-05-21:
@Jackman3323 The reset command does indeed fix the issue. 

From this, I found that the act of adding a depot via the CLI is what causes this bug. In my case it was LemLib latest stable: https://raw.githubusercontent.com/LemLib/LemLib/depot/stable.json. 

With this in mind I tried to recreate it on Windows and can confirm its only happening on my MacBook.

I'm not sure at this point if this is an issue with LemLib, the CLI depot functionality or the extension.

HY-Yap, 2025-05-21:
I have tried resetting the conductor already, yet I still face the same error when clicking the Create Project button. However, I am still able to create one with pros c new-project. Hope the issue can be fixed