purduesigbots/pros-vsc #230: 🐛

DylanBruner, 2024-09-09:
#### Bug Description

The extension doesn't work. The commands show in the VSC command pallet but whenever I try to run a command like "create new project" I get this,

[image]

#### Platform

Windows 11 ARM

#### Steps to reproduce

- Fresh install of VSC
- Install the PROS extension
- Try to run any PROS extension command

AndrewLuGit, 2024-09-09:
We don't currently support ARM Windows, but it looks like the extension is getting stuck somewhere before registering its commands with VSCode. Could you could you go to `%appdata%\Code\User\globalStorage\sigbots.pros\logs` and send the latest log file?

DylanBruner, 2024-09-09:
There's no `sigbots.pros` folder inside of globalStorage

AndrewLuGit, 2024-09-09:
Ok, it looks like it's a problem with node-usb which depends on a natively-built library libusb, and I don't think they are planning to support Windows on ARM soon. I'll look into what we can do to resolve this issue.

DylanBruner, 2024-09-10:
Ok, thanks!

AndrewLuGit, 2024-09-22:
Hey Dylan, I was able to get the extension to work on a Windows ARM virtual machine, but I wanted to make sure that it also works on your computer. Could you do the following?
1. Download [windows_arm_fix.zip](https://github.com/user-attachments/files/17090915/windows_arm_fix.zip)
2. Rename the file from `windows_arm_fix.zip` to `windows_arm_fix.vsix`
3. Use the vsix file to install the extension (see [this](https://code.visualstudio.com/docs/editor/extension-marketplace#_install-from-a-vsix))

Thanks!

DylanBruner, 2024-09-22:
It seems to be working! I was able to install the toolchain, create a project and build it. 
I'll update this once I get access to a v5 brain and can test uploading the program.

DylanBruner, 2024-09-23:
the main functionality seems to work, the only thing i've found that doesn't work is the battery medic, and the side menu for some reason? There doesn't seem to be any error messages in the logs?

It's not a big deal I just figured i'd let you know!

[image]
[image]

```
2024-09-23 13:02:23.220 [info] Extension host with pid 19156 started
2024-09-23 13:02:23.252 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2024-09-23 13:02:23.261 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2024-09-23 13:02:23.283 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2024-09-23 13:02:23.368 [info] Eager extensions activated
2024-09-23 13:02:23.387 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-09-23 13:02:23.390 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2024-09-23 13:02:23.564 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2024-09-23 13:02:40.856 [info] Extension host terminating: renderer closed the MessagePort
2024-09-23 13:02:40.869 [info] Extension host with pid 19156 exiting with code 0
2024-09-23 13:02:41.440 [info] Extension host with pid 12064 started
2024-09-23 13:02:41.440 [info] Skipping acquiring lock for c:\Users\brune\AppData\Roaming\Code\User\workspaceStorage\6bd12bb937a7ca6c966f5b38a494a4ab.
2024-09-23 13:02:41.500 [info] ExtensionService#_doActivateExtension vscode.git-base, startup: true, activationEvent: '*', root cause: vscode.git
2024-09-23 13:02:41.522 [info] ExtensionService#_doActivateExtension vscode.git, startup: true, activationEvent: '*'
2024-09-23 13:02:41.542 [info] ExtensionService#_doActivateExtension vscode.github, startup: true, activationEvent: '*'
2024-09-23 13:02:41.630 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2024-09-23 13:02:41.638 [info] Eager extensions activated
2024-09-23 13:02:41.693 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-09-23 13:02:41.696 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
2024-09-23 13:03:16.356 [info] ExtensionService#_doActivateExtension vscode.configuration-editing, startup: false, activationEvent: 'onLanguage:json'
2024-09-23 13:03:16.362 [info] ExtensionService#_doActivateExtension vscode.extension-editing, startup: false, activationEvent: 'onLanguage:json'
2024-09-23 13:03:16.370 [info] ExtensionService#_doActivateExtension vscode.json-language-features, startup: false, activationEvent: 'onLanguage:json'
2024-09-23 13:03:16.384 [info] ExtensionService#_doActivateExtension vscode.npm, startup: false, activationEvent: 'onLanguage:json'
2024-09-23 13:03:16.405 [info] ExtensionService#_doActivateExtension vscode.emmet, startup: false, activationEvent: 'onLanguage'
2024-09-23 13:03:21.354 [error] Error: No such branch: main.
	at N.getBranch (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:937115)
	at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
	at async T.retryRun (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1032816)
	at async T.run (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1032420)
	at async T.getBranch (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1020563)
	at async T.getBranchBase (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1020849)
	at async t.GitHistoryProvider.resolveHEADMergeBase (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:948410)
	at async t.GitHistoryProvider.onDidRunGitStatus (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:941737)
2024-09-23 13:03:21.467 [error] Error: No such branch: main.
	at N.getBranch (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:937115)
	at process.processTicksAndRejections (node:internal/process/task_queues:95:5)
	at async T.retryRun (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1032816)
	at async T.run (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1032420)
	at async T.getBranch (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1020563)
	at async T.getBranchBase (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:1020849)
	at async t.GitHistoryProvider.resolveHEADMergeBase (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:948410)
	at async t.GitHistoryProvider.onDidRunGitStatus (c:\Users\brune\AppData\Local\Programs\Microsoft VS Code\resources\app\extensions\git\dist\main.js:2:941737)
[... cut ...]
2024-09-23 18:11:55.817 [info] ExtensionService#_doActivateExtension vscode.github-authentication, startup: false, activationEvent: 'onAuthenticationRequest:github'
2024-09-23 18:11:56.162 [info] Eager extensions activated
2024-09-23 18:11:56.169 [info] ExtensionService#_doActivateExtension vscode.debug-auto-launch, startup: false, activationEvent: 'onStartupFinished'
2024-09-23 18:11:56.171 [info] ExtensionService#_doActivateExtension vscode.merge-conflict, startup: false, activationEvent: 'onStartupFinished'
```

AndrewLuGit, 2024-09-26:
I was able to replicate the battery medic issue, not sure what's up about your missing icon in the activity bar, though.

DylanBruner, 2024-09-26:
Let me know if you want me to test anything, but for my purposes it works just fine! Thanks for all your help :)