purduesigbots/pros-vsc #231: ✨Suggest Enabling Autosave

meisZWFLZ, 2024-09-21:
#### Requested Feature

Suggest that the user enable autosave, and provide buttons to turn it on or ignore it.
In either case, the user should not be prompted again. 
This can be done by writing to a pros setting called something like `dontSuggestAutosave`.

This would be best done when the user presses build/upload, by checking whether the code has been saved to disk, and if not, prompting the user to either save the file, enable autosave, or ignore.
I understand that would be much harder, so I think doing it on startup might work as well.

#### Is this Feature Related to a Problem?

Many of user's problems are due to them not understanding that the code they're changing is only in memory, and hasn't changed the actual code on disk, thats being compiled.
This leads to lots of confusion as to why the robot's behavior isn't changing.

#### Benefits of Feature

Implementing this would prevent the above problem, **as long as the user notices the prompt**.

#### Additional Information

BennyBot, 2024-09-22:
Going to hold off on this as it is a good first issue for recruits. The logic in the extension will look something like this, i think this is best:

Vscode PROS setting called `pros.promptautosave`, set on a per workspace setting (each PROS project), default set to true.
On extension activation, check:
If pros.promptautosave is true and the actual autosave setting is disabled, prompt the user to enable autosave with 3 buttons:
Enable, not now, never.

Enable will set vscode autosave mode to onfocuschange for that workspace/project,
Not now will just close dialogue, never will set pros.promptautosave to false and close dialogue.

onfocuschange works on clicking sidebar buttons and terminal, not sure about the build&upload quick access button (little pros logo)

Is this good @meisZWFLZ ?

meisZWFLZ, 2024-10-06:
Why on a per workspace basis?
Why `onfocuschange`? Pressing the pros logo button does not induce saving. I'd suggest `afterDelay`

Everything else looks good.

BennyBot, 2024-10-06:
Users may want autosave disabled in non-PROS projects, or maybe only enabled for certain PROS projects. Not sure why they would want this, but we should not change their global settings when prompting on a per project basis (PROS extension background activation is triggered by being in a PROS projects, not all vscode workspaces).

I think onFocusChange is the optimal setting, as it will be triggered just before the user clicks build or upload or anything non file editor related. Rn i think the PR has it set to afterDelay. I think on afterDelay there is some edge case of user making change and clicking build&upload before autosave kicks in, depending on the value of afterDelay.

meisZWFLZ, 2024-10-06:
I have never had problems with afterDelay, but I mostly use cli.

As for autosave being disabled in non-PROS projects, those users have two options:
- vscode profile for PROS projects
- enable autosave themselves in the workspace settings, which would disable the prompt

closed 2025-09-20 by AndrewLuGit