purduesigbots/pros-vsc #256: 🐛 PROS refusing to do anything

Just-Do-It-G, 2025-04-19:
PROS completely refuses to do anything. If it somehow does it, it takes forever. Sometimes it can't open the terminal, any terminal command takes at least 2+ minutes, sometimes close to 5 to execute, if it ever does so. Verifying PROS installation never stops loading. I can uninstall everything and reinstall it but it just doesn't work. For example, I uninstalled PROS and reinstalled everything except the CLI, it wouldn't download. Then I uninstalled the extension and did the same process, this time the toolchain didn't download. The program will not Build or Upload anything, it occasionally does but it take like 5 minutes to do so. 

Windows 11 (AMD Ryzen 7 PRO 7840U)

This isn't a bug per se but just an issue I've been having. I was told to ask here.

Just-Do-It-G, 2025-04-28:
After deleting every single PROS and VSCode file, I completely uninstalled and reinstalled VSCode. It started working but less than a week later, it 'broke' again. Verifying PROS installation doesn't work and keeps on loading, when it does finish, it says that either everything is up to date and working or that the CLI installation isn't working. Either message doesn't let me download and upload the program as it takes forever to build and upload.

Just-Do-It-G, 2025-06-08:
>My antivirus software (McAfee) then notifies me that it blocked a threat. It tells me the threat name is "hti!7d5a9a41" in the install\pros-cli-windows\pros.exe file. Is this an actual viral threat, or does Pros have an issue with its CLI compiles?

I use Cloudfare Warp and I've tried to disable it and download everything but it still fails to compile. Some of my teammates have it and it doesn't interrupt their building of the code. I doubt in your case it's an actual threat, its probably a false red flag but ignore at your own risk. 

Smt that worked for me is that I completely uninstalled all PROS and VSCode files on my computer, CLI, Toolchain, everything. I then tried to install everything and it started to work, PROS built like normal, the extension launched well, programs were being compiled fast. But then I quit VSCode, opened it up again and my problems started once more. PROS taking forever for the extension to launch and to compile the code. So irdk what to do.

istaawoo, 2025-10-01:
It was my built-in HP Wolf Security anti virus software that made everything run slow